chipsalliance/f4pga-examples #281: Symbiflow-classroom: Initialization of memories using $readmemh()

nelsobe, 2022-03-24:
SYMBIFLOW-CLASSROOM-PROJECT

Using Yosys front end.

The following code does NOT initialize the memory:

		// Load the Instruction Memory
		if (TEXT_MEMORY_FILENAME == "") begin
			$display("**** Top-Level I/O System: No instruction memory defined");
			$finish;
		end
		else begin
			// Initialize memory with NOPs
			for (i = 0; i < INSTRUCTION_WORDS; i=i+1)
				inst_memory[i] = NOP_INSTRUCTION;
			// Update memory with contents of memory file
        	$readmemh(TEXT_MEMORY_FILENAME,inst_memory);
		end

Experimentation shows that the testing of the TEXT_MEMORY_FILENAME works just fine.  The problem turns out to be the for loop which initializes the memory to NOP_INSTRUCTION values before the #readmemh().   Commenting out the for loop makes everything work.   However this is code that works just fine with Vivado.

nelsobe, 2022-03-29:
@rkapuscik 

I am unsure how to get the latest surelog with the f4pga-examples tools - thus my question on #269.  

However, attached is a full test case up through bitfile showing that the $readmem is ignored when the initialization loop is included.  This is for the Yosys front end.

[readmem.zip](https://github.com/chipsalliance/f4pga-examples/files/8373371/readmem.zip)

rkapuscik, 2022-04-08:
Thanks for the test case. The error is indeed in Yosys itself - both frontends parse the loop and `$readmemh` call and generate the AST, but the simplify step in Yosys incorrectly removes some of the nodes. Unfortunately it's not trivial (if possible) to prevent that in a frontend.

As a workaround, could you create a complete memory file to be loaded (with the NOPs)? It should be fairly easy to script.

nelsobe, 2022-04-08:
Thanks for the reply.  Yes, we did do a workaround as you suggest.  Interestingly, I never would have thought to put anything in an initial block other than a $readmem() call.  But, one of the other faculty did - that is why we are having lots of people's designs be part of our testing...