chipsalliance/f4pga-examples #288: SYMBIFLOW-CLASSROOM UHDM doesn't handle typed parameters

nelsobe, 2022-04-12:
The following code throws a compile error on the parameter statement when compiled with the UHDM.  Removing the 'integer' solves it.    It compiles and runs correctly with both Vivado as well as the original Yosys front end.
```
module top(
    input logic[3:0] sw,
    output logic [3:0] led
);

parameter integer OFFSET=5;

assign led = sw + OFFSET;

endmodule
```

A complete test case is below.  The command line used to compile with the latest version of f4pga-examples is:
```
SURELOG_CMD="-parse -DSYNTHESIS" make
```
Note that my Makefile sets the part itself and so that is not included.

[Issue282.zip](https://github.com/chipsalliance/f4pga-examples/files/8476812/Issue282.zip)

jcrob2, 2022-06-17:
Using the following timestamp and hash, the issue still exists.
Timestamp: 20220606-234655
Hash: 1667c14

I've included a file that contains a list of the content/versions of the conda environment that we have.
[conda_list.txt](https://github.com/chipsalliance/f4pga-examples/files/8931052/conda_list.txt)

Zach227, 2022-07-20:
@nelsobe I have retested this bug with the latest install and have verified that it is fixed. This issue can be closed.

closed 2022-07-20 by nelsobe