chipsalliance/f4pga-examples #289: SYMBIFLOW-CLASSROOM Surelog/UHDM Simple Combinational Logic Design Doesn't Work

nelsobe, 2022-04-12:
The attached design results in a non-functional .bit file.

Would be interested in knowing how to dump an AST to see what is coming out to try to find where the issue is.

However, design does work in Vivado and with the Yosys front end.

Tools run using `SURELOG_CMD="-parse -DSYNTHESIS" make`

[Issue289.zip](https://github.com/chipsalliance/f4pga-examples/files/8476874/Issue289.zip)

rkapuscik, 2022-04-15:
Thanks for reporting this, we haven't handled built-in gates so far. https://github.com/chipsalliance/yosys-f4pga-plugins/pull/302 adds support for them and with it your designs give the same AST as Yosys' default frontend.

nelsobe, 2022-05-09:
@rkapuscik @mithro @acomodi @mkurc-ant @tmichalak Thanks to @rkapuscik for providing the fix for this.  

However, when a fix is made for an issue such as this, we currently have no way of knowing when the fix has made its way through the rest of CI, the f4pga-arch-defs process, and into the f4pga-examples documentation so that the correct packages get installed by our student testers.  Without this knowledge our students have no way of knowing when they can re-test so the f4pga-examples issue can be closed.   

How can we determine when f4pga-examples contains fixes such as this one?

mithro, 2022-05-10:
@mgielda / @pgielda / @kgugala - Can you please look at this and figure out what we should be doing about this?

pgielda, 2022-05-10:
So essentially f4pga-examples installs packets via conda with versions (for xc7) specified by this file:
https://github.com/chipsalliance/f4pga-examples/blob/main/xc7/environment.yml
This change is in plugins so at the time I am writing f4pga-examples use this package:
litex-hub::symbiflow-yosys-plugins=1.0.0_7_855_gd64a754=20220330_123635
which is dated for 30th of March.
The change was merged around 2 weeks later so this is not yet present in the examples as of the time I am writing this.

Looking at the packages available:
https://anaconda.org/LiteX-Hub/symbiflow-yosys-plugins/files
seems that there are newer packages available.

@kgugala can you describe the procedure on how/when f4pga-examples environment.yml is updated?

kgugala, 2022-05-10:
we update the file manually - to be sure the examples are stable. I'll open a PR bumping the plugins version

nelsobe, 2022-05-10:
@kgugala @mithro Thanks for the response. 

If a process could be agreed upon so that our students could be informed of just what modules to use with f4pga-examples to test a specific bug fix once it is made, they could be a part of the verification process and help provide data on  when the version can be bumped.   

Without it, every issue we file requires this kind of personal interaction to get a distribution our students can test against.  I think our students could help much more and progressm could accelerate if we can get the process correct.   Thoughts?

jcrob2, 2022-06-17:
Using the following timestamp and hash, the issue still exists.
Timestamp: 20220606-234655
Hash: 1667c14

I've included a file that contains a list of the content/versions of the conda environment that we have.
[conda_list.txt](https://github.com/chipsalliance/f4pga-examples/files/8931052/conda_list.txt)

rkapuscik, 2022-06-30:
This should be already fixed, please update the plugin to the newest version as described [here](https://github.com/chipsalliance/f4pga-examples/blob/main/docs/getting.rst#updating-yosys-systemverilog-plugin) to verify.

Zach227, 2022-07-20:
@nelsobe I have retested this bug with the latest install and have verified that it is fixed. This issue can be closed.

nelsobe, 2022-07-20:
Thanks @Zach227.  Closing this issue.

closed 2022-07-20 by nelsobe