chipsalliance/f4pga-examples #290: SYMBIFLOW-CLASSROOM Surelog/UHDM Doesn't Handle `default_nettype none

nelsobe, 2022-04-13:
The macro `default_nettype none` is commonly used to cause compilers to throw an error when an undeclared net name is encountered since the tools cannot infer a new net of a given type with this macro in place.  It catches many typo errors as a result.

Here is a sample design that should result in an error :
```
`default_nettype none
module test(input wire logic a, output logic outwire);
    assign outwre = ~a;
endmodule
```
Both Vivado and the Yosys front end handle it correctly (they throw an error).  The Surelog/UHDM front end happily generates a bitfile without error (but which doesn't work).  

Here is the Yosys front end error, which is what one would expect:
```
7. Executing AST frontend in derive mode using pre-parsed AST for module `\test'.
Generating RTLIL representation for module `\test'.
/home/nelson/220-nelsobe/Labs/defaultnettype/test.sv:3: ERROR: Identifier `\outwre' is implicitly declared and `default_nettype is set to none.
make: *** [/home/nelson/f4pga-examples/common/common.mk:61: /home/nelson/220-nelsobe/Labs/defaultnettype/build/basys3/test.eblif] Error 1
```

rkapuscik, 2022-04-28:
Thanks for reporting this, https://github.com/chipsalliance/yosys-f4pga-plugins/pull/301 should make the frontend error out in this case. I will let you know when the fix will be available through conda.

jcrob2, 2022-06-17:
Using the following timestamp and hash, the issue still exists.
Timestamp: 20220606-234655
Hash: 1667c14

I've included a file that contains a list of the content/versions of the conda environment that we have.
[conda_list.txt](https://github.com/chipsalliance/f4pga-examples/files/8931052/conda_list.txt)

rkapuscik, 2022-06-30:
This should be already fixed, please update the plugin to the newest version as described [here](https://github.com/chipsalliance/f4pga-examples/blob/main/docs/getting.rst#updating-yosys-systemverilog-plugin) to verify.

Zach227, 2022-07-20:
@nelsobe I have retested this bug with the latest install and can verify that it is fixed. This issue can be  closed.

closed 2022-07-20 by nelsobe