chipsalliance/f4pga-examples #298: Convert examples to use `f4pga build` instead of makefiles

umarcor, 2022-05-18:
Since https://github.com/chipsalliance/f4pga/pull/530 was merged, the examples in this repository can be converted to use `f4pga build` instead of the current makefile based infrastructure. Python package `f4pga` includes entrypoints for the *deprecated* wrappers along with the new `build` subcommand. Therefore, examples can be converted one-by-one.

However, we need to first address #297.

mithro, 2022-05-20:
@umarcor - We need to make sure that existing Makefile based flows continue to work.

umarcor, 2022-05-20:
@mithro, see #297 and https://github.com/SymbiFlow/f4pga-arch-defs/pull/2509#issuecomment-1131394460.

mithro, 2022-05-20:
@umarcor - I'm not sure I understand what is relevant in #297 and https://github.com/SymbiFlow/f4pga-arch-defs/pull/2509#issuecomment-1131394460 to my comment?

umarcor, 2022-05-20:
@mithro, existing shell wrappers were moved from repo arch-defs to f4pga. Previously, they were installed as part of the arch-defs packages. Now, they need to be installed through pip, and 1 or 2 envvars need to be set (mentioned in https://github.com/SymbiFlow/f4pga-arch-defs/pull/2509#issuecomment-1131394460). Other than that, the Makefile based flows will continue to work because the shell wrappers are exactly the same (just moved and installed using a different procedure).

#297 is to track updating the guidelines in this repo in order to install the pip package and set the envvars, but without modifying any makefile at all.

On top of that, the pip package does include a new Python-only implementation, as an alternative to the makefile/shell based approach. This issue is to track updating the examples in this repo to use the Python-only approach.

mithro, 2022-05-20:
@umarcor - As long as people's existing Makefile based flows continue to work (and are ideally also being tested in some way) then I'm happy.

(PS There is no reason that Python scripts can't be drop in 100% compatible replacement for shell scripts?)

umarcor, 2022-05-20:
> (PS There is no reason that Python scripts can't be drop in 100% compatible replacement for shell scripts?)

@mithro, there was a prototype (https://github.com/chipsalliance/f4pga/commit/2291026181192b604e5eb44b29113c47b055f118), but the current implementation is based on a JSON configuration file (https://github.com/chipsalliance/f4pga/blob/main/.github/sftest.json). It should be possible to replace the *deprecated* wrapper commands with functions calling one specific internal module/step/stage of f4pga. That's why the entrypoints are generated in Python, despite calling shell scripts internally, so that future changes in the implementation are "transparent". It is desirable to make internal modules/stages usable from Python, indeed, so that other projects can better integrate with f4pga (say edalize, EDAA, Xeda, etc.). However, I am not aware of the technical details and discussions that lead to the current implementation of command `build`, and I'm not familiar enough with the codebase yet.

umarcor, 2022-06-09:
FTR, example xc7/counter_test was modified to use `f4pga build` when targeting arty_35: #318.