chipsalliance/f4pga-examples #302: Couldn't read file /scripts/xc7/synth.tcl

Pocketkid2, 2022-05-19:
Installing the F4PGA flow and attempting to build the counter example according to the instructions on the pages [here](https://f4pga-examples.readthedocs.io/en/latest/getting.html) and [here](https://f4pga-examples.readthedocs.io/en/latest/getting.html) results in the following error:

[log.txt](https://github.com/chipsalliance/f4pga-examples/files/8732823/log.txt)

I had previously installed F4PGA but I do not understand what has changed or how I can fix this without editing the scripts.

umarcor, 2022-05-19:
@Pocketkid2 can you please explain the procedure you followed? Did you install f4pga through pip? Did you do it after or before creating the Conda environment?

Pocketkid2, 2022-05-19:
I followed the instructions on the first page linked above which involved setting up the environment and then running the wget commands to download the code. Then I went and ran the make command shown on the second linked page.

tcal-x, 2022-05-19:
@Pocketkid2 can you check that you have `INSTALL_DIR` set correctly when trying to build?

umarcor, 2022-05-19:
@Pocketkid2 didn't you install Python package f4pga through pip? Maybe not today but yesterday or some other day? I'm asking because the log shows the following:

```
Traceback (most recent call last):
  File "~/opt/f4pga/xc7/conda/envs/xc7/bin/symbiflow_synth", line 8, in <module>
    sys.exit(synth())
  File "~/opt/f4pga/xc7/conda/envs/xc7/lib/python3.7/site-packages/f4pga/wrappers/sh/__init__.py", line 60, in synth
    run_sh(ROOT / SH_SUBDIR / "synth.f4pga.sh")
....
```

The current guidelines in f4pga-example do not have any reference to installing the Python package. Therefore, I wonder how did that end up in your environment... Note that both links in your first message seem to be pointing to the same place. Maybe that's why I'm not getting the full picture.

To be clear, as commented in https://github.com/SymbiFlow/f4pga-arch-defs/pull/2509#issuecomment-1131394460, we are transitioning from shell wrappers to python utils. Hence, there might be bugs/issues indeed.

umarcor, 2022-05-19:
@tcal-x, FYI, the shell wrappers were removed from the arch-defs packages and do now need to be installed through `pip install https://github.com/chipsalliance/f4pga/archive/main.zip#subdirectory=f4pga`. That will create some entrypoints through Python with the same names as the *deprecated* wrappers, which will call the now internal (*hidden*) shell wrappers. The guidelines in f4pga-examples need to be updated to explain that two environment variables need to be set before installing f4pga through pip (see https://github.com/SymbiFlow/f4pga-arch-defs/issues/2669). However, since the guidelines in f4pga-examples were not updated yet, users following them should be unaware and unaffected by this transition. I think this issue is indirectly related to chipsalliance/f4pga#557.

Pocketkid2, 2022-05-19:
I did install the python package for testing separately, but that had its own issues so I followed along with the f4pga-examples to see if that method worked. And I did verify that my INSTALL_DIR is correct, I just edited the log file to replace my full home path with ~

umarcor, 2022-05-19:
@Pocketkid2 if the content of `~/opt/f4pga/xc7/conda/envs/xc7/bin/symbiflow_synth` is *not* a shell script, you need to execute `wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/20220406-185509/symbiflow-arch-defs-install-3ef4188.tar.xz | tar -xJC $INSTALL_DIR/xc7/install` again. That should ovewrite the *new* entrypoints with the *old* ones (expected by f4pga-examples at the moment).

Pocketkid2, 2022-05-20:
@umarcor I have run that wget command along with the others but it doesn't seem to be overwriting the scripts. Is there something else I need to do? I do realize that this method of running symbiflow is ultimately meant to be replaced with the python module so I might close this issue and focus on debugging and reporting issues for that.

umarcor, 2022-05-20:
@Pocketkid2 do you find yourself comfortable for managing multiple environments/setups in parallel? If that is the case, I propose to keep the current one as is (for debugging/fixing this issue). Then create two new, one for examples as default, and another one for examples along with Python f4pga. In order to create a new environment, changing the INSTALL_DIR and running should suffice.
If you cannot (maybe because of disk size), let's focus on the issue you prefer only.

Pocketkid2, 2022-05-20:
I think for now I'll just focus on the python module, as that is more relevant to my current research project, and I have been meaning to dive into the code for that anyway. Thanks for your comments and help! I will be using the other issue I created on the main f4pga repo going forward.

closed 2022-05-20 by Pocketkid2