chipsalliance/f4pga-examples #309: Flow continues with Undefined Type Error using Surelog

westonMS, 2022-05-31:
## Issue
When declaring an invalid type the Surelog parser gives the following error: 
`[ERR:CP0317] {PATH}/top.sv:2: Undefined type "param".`

Despite this error being identified, the flow continues to the end until a bitstream is generated. 

The default yosys parser catches this and gives: `ERROR: syntax error, unexpected '=', expecting '(' or '['`.

## Code Example
```
module top(output logic outLED);
sandwich TEST_PARAM = "TestParam.txt";
assign outLED = 1;
endmodule
```

rkapuscik, 2022-06-03:
This should be already fixed with https://github.com/chipsalliance/yosys-f4pga-plugins/pull/301, but it seems that the tools in this repository were not updated yet.

jcrob2, 2022-06-17:
Using the following timestamp and hash, the issue still exists.
Timestamp: 20220606-234655
Hash: 1667c14

I've included a file that contains a list of the content/versions of the conda environment that we have.
[conda_list.txt](https://github.com/chipsalliance/f4pga-examples/files/8931052/conda_list.txt)

rkapuscik, 2022-06-30:
This should be already fixed, please update the plugin to the newest version as described [here](https://github.com/chipsalliance/f4pga-examples/blob/main/docs/getting.rst#updating-yosys-systemverilog-plugin) to verify.