chipsalliance/f4pga-examples #310: String Parameters not supported with Surelog-yosys Plugin

westonMS, 2022-05-31:
## Issue
If you use `localparam` or `parameter` to initialize parameters, Surelog-yosys plugin throws an error if you initialize it with a string. 
The yosys parser is able to correctly interpret this. Both can interpret the parameter correctly if you initialize it with an integer.

Run with command `SURELOG_CMD="-parse -DSYNTHESIS" make`
## Code Example
```
module top(output logic outLED);

localparam TEST_PARAM = "TestParam.txt";

assign outLED = 1;

endmodule
```
This code gives the following error:
`ERROR: :0: Encountered unhandled typespec in process_parameter: '' of type 'string_typespec'`

The Surelog Parser does not throw an error and handles it correctly.

rkapuscik, 2022-06-03:
Should be already fixed with https://github.com/chipsalliance/yosys-f4pga-plugins/pull/290, but it seems that the tools in this repository were not updated yet.

jcrob2, 2022-06-17:
Using the following timestamp and hash, the issue still exists.
Timestamp: 20220606-234655
Hash: 1667c14

I've included a file that contains a list of the content/versions of the conda environment that we have.
[conda_list.txt](https://github.com/chipsalliance/f4pga-examples/files/8931052/conda_list.txt)

rkapuscik, 2022-06-30:
This should be already fixed, please update the plugin to the newest version as described [here](https://github.com/chipsalliance/f4pga-examples/blob/main/docs/getting.rst#updating-yosys-systemverilog-plugin) to verify.

westonMS, 2022-07-14:
I've retested this with the latest install and have verified that it works

closed 2022-07-14 by westonMS