chipsalliance/f4pga-examples #311: Typedef Enum Width Default to one bit

westonMS, 2022-05-31:
Using the Surelog parser,  when initializing an enum, if a width is not set it defaults to one bit.
The bitstream is generated but is faulty.

Works:
```
    typedef enum {S_INIT, S_LOOP1,
                    S_LOOP2_readSi, S_LOOP2_readSj, S_LOOP2_write,
                    S_LOOP3_init, S_LOOP3_readSi, S
[decrypt_rc4.zip](https://github.com/chipsalliance/f4pga-examples/files/8808581/decrypt_rc4.zip)
_LOOP3_readSj, S_LOOP3_writeSi, S_LOOP3_writeSj, S_LOOP3_readK,
                    S_update_text_out,
                    S_DONE} StateType;
    StateType [4:0] cs;
```

Original does not work:
[decrypt_rc4.zip](https://github.com/chipsalliance/f4pga-examples/files/8808580/decrypt_rc4.zip)
```
    typedef enum {S_INIT, S_LOOP1,
                    S_LOOP2_readSi, S_LOOP2_readSj, S_LOOP2_write,
                    S_LOOP3_init, S_LOOP3_readSi, S_LOOP3_readSj, S_LOOP3_writeSi, S_LOOP3_writeSj, S_LOOP3_readK,
                    S_update_text_out,
                    S_DONE} StateType;
    StateType cs;
```

This was run with the `SURELOG_CMD='-parse -DSYNTHESIS' make`

Error did not occur running with `make`

westonMS, 2022-05-31:
[Same issue](https://github.com/chipsalliance/f4pga-examples/issues/291#issue-1203841388) from earlier with a different fix.

rkapuscik, 2022-06-03:
Thanks for reporting, I reproduced this and we'll look into it.

rkapuscik, 2022-06-30:
This should be already fixed, please update the plugin to the newest version as described [here](https://github.com/chipsalliance/f4pga-examples/blob/main/docs/getting.rst#updating-yosys-systemverilog-plugin) to verify.

westonMS, 2022-07-14:
We have re-tested this with the latest plugin and the design works now. However, we want fixes to be available to all users when they use the main install. I re-installed the tool-chain with the latest install instructions and this is not fixed without manually updating the plugin.

mithro, 2022-07-16:
FYI - @kgugala

closed 2022-07-26 by westonMS