chipsalliance/f4pga-examples #313: Signal not in Net List with one bit wires

westonMS, 2022-05-31:
When referencing a single bit wire, defined using `[0:0]` as its width, in an XDC file using `[0]` (e.g. `sw[0]`) the following error is returned:
```
ERROR:
Constrained net sw[0] is not in available netlist:
sw
led
```

This is the problematic line: `  set_property -dict { PACKAGE_PIN V17   IOSTANDARD LVCMOS33 } [get_ports { sw[0] }];`

The following does work with Vivado using `input wire logic [0:0] sw` but does not work with `input wire logic sw`
[broken_wire_dec (1).zip](https://github.com/chipsalliance/f4pga-examples/files/8809377/broken_wire_dec.1.zip)

rkapuscik, 2022-06-03:
Thanks for reporting, we will investigate this.

rkapuscik, 2022-07-05:
Note: UHDM is exactly the same for both `input wire logic [0:0] sw` and `input wire logic sw`, so it cannot be easily fixed in SystemVerilog plugin.

wirthlin, 2022-07-27:
This seems to be specific to how Xilinx handles xdc files and so we are not going to worry about it.

closed 2022-07-27 by wirthlin