chipsalliance/f4pga-examples #316: Codebreaker Lab Failure

Zach227, 2022-06-06:
When running the Codebreaker Lab through the f4pga flow with the default Yosys parser it works correctly. When the Surelog parser is used, it generates a bit stream but does not function properly on the board. The seven segment display works but it does not decrypt the message.

Full Project: [codebreaker.zip](https://github.com/chipsalliance/f4pga-examples/files/8847109/codebreaker.zip) (not working with Surelog)

Replace decrypt_rc4.sv from the project with this [edited version](https://github.com/chipsalliance/f4pga-examples/files/8847127/edited_decrypt_rc4.zip) and the project will work with Surelog. There are comments on lines 38 and 121 pointing out the 2 changes to the file. When working properly, the programmed FPGA will decrypt the message and stop the timer at 11.88 seconds. It will print the message to a UART receiver (19200 baud, 8 data bits, 1 stop bit, odd parity).

rkapuscik, 2022-06-07:
Thanks for the report.

One of the changes (explicit width of enum variable) is I think a duplicate of #311 (it seems to be the same test case?). I'm not sure about the second change (merging the `always_ff`), we'll look into that.

rkapuscik, 2022-07-05:
I've retested this and after the enum variable fix the Yosys tree is functionally the same with or without merging the `always_ff` block, so no changes should be needed anymore.

Please retest with the latest plugin.

mithro, 2022-07-16:
FYI - @kgugala

Zach227, 2022-07-26:
I have retested with the latest install instructions and this issue is resolved.

closed 2022-07-26 by Zach227