chipsalliance/f4pga-examples #364: make: f4pga: No such file or directory make: *** [Makefile:10: build] Error 127

YCin, 2022-09-28:
Hi 

when i try to build a file in rpi4 platform i have the following error

make: f4pga: No such file or directory
make: *** [Makefile:10: build] Error 127

any suggestion or solutiThanks

umarcor, 2022-09-28:
Please, provide details about the procedure you followed. Running make without first creating the Conda environment is likely to fail.

pgielda, 2022-09-28:
Also I am pretty sure f4pga-examples will not work on RPi4 as we do not currently provide ARM packages on conda (yet)...

YCin, 2022-10-01:
> Also I am pretty sure f4pga-examples will not work on RPi4 as we do not currently provide ARM packages on conda (yet)...

Have you Know any other solution to make f4pga-examples on RPI4.
Thanks

YCin, 2022-10-01:
> Please, provide details about the procedure you followed. Running make without first creating the Conda environment is likely to fail.

The project consist on impliment verilog code with openfpga bootloader by the mean of rpi4