chipsalliance/f4pga-examples #366: Built in gates cannot take multi-bit signal (Surelog)

westonMS, 2022-10-07:
Using the Surelog parser, built in gates (and, or, not, xor etc.) cannot take multi bit signals as inputs or outputs. If a multi-bit signal is used (e.g. sw[5]) it will default to the first bit instead (sw[0]).

Example:
[gate_test.zip](https://github.com/chipsalliance/f4pga-examples/files/9737714/gate_test.zip)

umarcor, 2022-10-09:
Is this related to #269 ?

westonMS, 2022-10-19:
From testing, it appears that issue #269 is still fixed and not the problem.

rkapuscik, 2022-10-21:
Thanks for reporting this. I tried to reproduce it, but the Verilog code emitted after parsing through plugin seems correct:
```
assign I = (N[0:0])&(N[1:1]);
assign J = (N[2:2])|(N[3:3]);
/** AST_GENBLOCK **/
assign O[0:0] = (N[5:5])^(N[6:6]);
(* always_comb = 1 *)
always @*
  begin
    N[0:0] = sw[0:0];
    N[1:1] = sw[1:1];
    N[2:2] = sw[2:2];
    N[3:3] = sw[3:3];
    N[4:4] = sw[4:4];
    N[5:5] = sw[5:5];
    N[6:6] = sw[6:6];
    O[0:0] = led[0:0];
    I = led[1:1];
    J = led[2:2];
    K = led[3:3];
  end
assign K = ~(N[4:4]);
```
Yosys parser produces the same assignments. Is this design expected to work with Yosys parser? How do you verify that wrong bit was used?

wsipak, 2023-04-06:
With Yosys 0.27+22 `(git sha1 0f5e7c244)` and yosys-f4pga-plugins `(git sha1 e7070ca645)`, which are used here: https://github.com/antmicro/yosys-systemverilog I don't see any problem here. I think the issue can be closed.