chipsalliance/f4pga-examples #378: The final installation command is very easy to mess up

JamesTimothyMeech, 2023-05-18:
The final installation command is very easy to mess up. When I paste it into my terminal it does download and unzip the packages into the xc7 install directory as it should
`for PKG in $F4PGA_PACKAGES; do
  wget -qO- https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/${F4PGA_TIMESTAMP}/symbiflow-arch-defs-${PKG}-${F4PGA_HASH}.tar.xz | tar -xJC $F4PGA_INSTALL_DIR/${FPGA_FAM}
done`

I get this error when trying to run the examples so I assume that my install did not work correctly:
`meechy@DESKTOP-5HD2OJR:~/Casino/f4pga-examples/xc7$ TARGET="arty_100" make -C counter_test
make: Entering directory '/home/meechy/Casino/f4pga-examples/xc7/counter_test'
mkdir -p /home/meechy/Casino/f4pga-examples/xc7/counter_test/build/arty_100
cd /home/meechy/Casino/f4pga-examples/xc7/counter_test/build/arty_100 && symbiflow_synth -t top  -v /home/meechy/Casino/f4pga-examples/xc7/counter_test/counter.v -d artix7 -p xc7a100tcsg324-1 -x /home/meechy/Casino/f4pga-examples/xc7/counter_test/arty.xdc
/bin/sh: 1: symbiflow_synth: not found
make: *** [/home/meechy/Casino/f4pga-examples/xc7/counter_test/../../common/common.mk:67: /home/meechy/Casino/f4pga-examples/xc7/counter_test/build/arty_100/top.eblif] Error 127
make: Leaving directory '/home/meechy/Casino/f4pga-examples/xc7/counter_test'`

pgielda, 2023-05-19:
You would  have to paste a detailed log from start to finish, otherwise not really possible to debug. There is a CI that runs those instructions and it runs them successfully on a lot of distributions, so technically the instructions are correct. But a lot of stuff is downloaded from the internet, if anything failed then obviously you will end up with a broken install. This is just a guess, but I would just try to go through the instructions again from scratch, maybe it was a glitch.

JamesTimothyMeech, 2023-05-19:
You were right, the first time around I pasted the last command as two separate pieces which for some reason stopped it from working even though I pasted the whole for loop in one go

closed 2023-05-19 by JamesTimothyMeech